Microchip-MPLAB-Harmony/csp #2: ANSEL in GPIO Initialize

edndg, 2019-05-06:
ANSEL is not present on Port C of PIC32MZ2048EFM064 but it was generated.

aethaniel, 2019-05-06:
Issue transfered to its right place. 
@edndg, thanks for reporting. We are checking it.

aethaniel, 2019-05-06:
@edndg, the issue has been fixed recently and the coming next release will show as you expect.

closed 2019-05-06 by aethaniel

youpko, 2020-01-24:
This problem still persists for the PIC32MZ1024EFK064

arpananand1, 2020-01-27:
> This problem still persists for the PIC32MZ1024EFK064

can you tell the steps to reproduce the issue you are facing?

youpko, 2020-01-27:
Generate source files for the PIC32MZ1024EFK064
in _plib_gpio.c_ in _GPIO_Initialize_ there is generated code for ANSELCCLR register, but the PIC in question doesn't have analog capabilities on Port C.

arpananand1, 2020-01-28:
> Generate source files for the PIC32MZ1024EFK064
> in _plib_gpio.c_ in _GPIO_Initialize_ there is generated code for ANSELCCLR register, but the PIC in question doesn't have analog capabilities on Port C.

There is a new release planned very soon, you should see the issue fixed in that release.