Microchip-MPLAB-Harmony/csp #3: dfp path  is not  valid     in ubuntu14.04

Kingders, 2019-05-20:
how do i fix this ,  when i just open a  sama5d2-demo  followed by  Get Started with Harmony 3 on the SAMA5D2 pages,  i confirmed  my  DFP and CMSIS path  placed in win-form "configuration database setup ",  i confirmed  read-write-exec priority of the path and files, but not worked , and the Launch button is always gray !!

djsaz, 2019-05-20:
Can you please show me the command you used to launch MHC?  Do you see any red error messages saying a path is invalid?

Kingders, 2019-05-21:
my command  :  java -jar ./mhc/mhc.jar -fw="/home/userA//MPLAB_XIDE/harmonyFrameworkv3" -mode=gui -c="/home/userA/MPLAB_XIDE/harmonyFrameworkv3/csp/apps/uart/uart_echo_interrupt/firmware/src/config/sam_a5d2_xult/harmony.prj"   

then, come out a window called "Configuration database setup", 
 red error messages:" DFP path is not valid."  in the bottom of the window
DFP  ( auto-filled )  :  ./dev_packs/Microchip/SAMA5D2_DFP/0.1.6/atdf/ATSAMA5D27.atdf
CMSIS  ( auto-filled)      ./dev_packs/arm/CMSIS/5.4.0
and the bottom  "Launch"  is  GRAY, (mean disabled, can't click)

though, i change  DFP  CMSIS  input  in  whole path !!
DFP:     /home/userA/MPLAB_XIDE/harmonyFrameworkv3/dev_packs/Microchip/SAMA5D2_DFP/0.1.6/atdf/ATSAMA5D27.atdf
CMSIS :    /home/userA/MPLAB_XIDE/harmonyFrameworkv3/dev_packs/arm/CMSIS/5.4.0
 red error messages  still  be  here
and the bottom  "Launch"  still  be  GRAY,

by the way, i run in  ubuntu14.04 ,  and  i just only be applied to  use  this machine  , no other PC in window system can be used

Kingders, 2019-05-21:
my command : java -jar ./mhc/mhc.jar -fw="/home/userA//MPLAB_XIDE/harmonyFrameworkv3" -mode=gui -c="/home/userA/MPLAB_XIDE/harmonyFrameworkv3/csp/apps/uart/uart_echo_interrupt/firmware/src/config/sam_a5d2_xult/harmony.prj"

then, come out a window called "Configuration database setup",
red error messages:" DFP path is not valid." in the bottom of the window
DFP ( auto-filled ) : ./dev_packs/Microchip/SAMA5D2_DFP/0.1.6/atdf/ATSAMA5D27.atdf
CMSIS ( auto-filled) ./dev_packs/arm/CMSIS/5.4.0
and the button "Launch" is GRAY, (mean disabled, can't click)

though, i change DFP CMSIS input in whole path !!
DFP: /home/userA/MPLAB_XIDE/harmonyFrameworkv3/dev_packs/Microchip/SAMA5D2_DFP/0.1.6/atdf/ATSAMA5D27.atdf
CMSIS : /home/userA/MPLAB_XIDE/harmonyFrameworkv3/dev_packs/arm/CMSIS/5.4.0
red error messages still be here
and the button "Launch" still be GRAY,

by the way, i run in ubuntu14.04 , and i just only be applied to use this machine , no other PC in window system can be used

djsaz, 2019-05-21:
Two things, 1) make sure you run the MHC that is in the /home/userA/MPLAB_XIDE/harmonyFrameworkv3/mhc directory.  2) You have to run mhc from the mhc directory.  You can't run:
java -jar ./mhc/mhc.jar 
You must cd to mhc then run:
java -jar ./mhc.jar

Kingders, 2019-05-22:
i know what you mean,

now  my folder's  tree like this: 
$ ls  harmonyFrameworkv3
audio  bt    core    csp     gfx    micrium_ucos3   net    touch  wolfssl
bsp    CMSIS-FreeRTOS   crypto  dev_packs  gfx_apps  mhc   motor_control  project   usb

$ ls   harmonyFrameworkv3/mhc
mhc .jar  ......

then  i  run   mhc.jar   in  folder  mhc :
java -jar ./mhc.jar -fw="/home/userA/MPLAB_XIDE/harmonyFrameworkv3" -mode=gui -c="/home/userA/MPLAB_XIDE/harmonyFrameworkv3/csp/apps/uart/uart_echo_interrupt/firmware/src/config/sam_a5d2_xult/harmony.prj"

but  not  work,  problem still  exits

Kingders, 2019-05-22:
project just is my workfolder,   not the one harmonyV3  provided

djsaz, 2019-05-22:
You  need the trailing / in your fw path.  It should be 
-fw=/home/userA/MPLAB_XIDE/harmonyFrameworkv3/

Kingders, 2019-05-23:
as you see
my command use trailing  /  always!!!
so these may not be the point causing this problem,,,,,,,

though,  i tranfer
-fw="/home/userA/MPLAB_XIDE/harmonyFrameworkv3/"
 to
-fw=/home/userA/MPLAB_XIDE/harmonyFrameworkv3/

problem still exits

djsaz, 2019-05-23:
With H3 downloaded and with the commands formatted as I've suggested, I am unable to reproduce what you're seeing.  Would you be willing to do a webex meeting where I can see your screen and hopefully see what's missing?

closed 2019-08-09 by djsaz