Microchip-MPLAB-Harmony/csp #4: Configuration for ATSAME70N20 failes for PIO and CLK

macload1, 2019-06-13:
The Harmony configurator doesn't correctly configure the register UTMI_REGS->UTMI_CKTRIM in file "plib_clk.c". In fact, either configuring for a 12 MHz or a 16 MHz crystal in the harmony configurator, the register is always set to UTMI_CKTRIM_FREQ_XTAL12 and never to UTMI_CKTRIM_FREQ_XTAL16. This is maybe a problem in the configurator and not in the template files.
Second problem is for PIO. The device doesn't have a PORTC but has a PORTD. The configurator fails first to execute. The fix for this is to change the line 657 of the file "plib_pio.c.ftl" from <#if PIO_C_INTERRUPT_USED??> to <#if PIO_D_INTERRUPT_USED??>.
Then, the generated plib_pio.c file has a configuration for PORTC but not for PORTD, so the compilation fails. You need to manually delete all PORTC related registers and configure PORTD, but all PORTD related configuration from the visual pin manager is lost.

macload1, 2019-08-20:
I suppose that this [commit](https://github.com/Microchip-MPLAB-Harmony/csp/commit/d2842ffc821f05728823dff855a136db44e72a8b) fixed the PORTD problem.
For the crystal problem, I didn't find the suitable commit yet.
It would be nice if the Microchip team could comment the issues and their resolution here.
Many thanks in advance.

freaX22, 2019-08-25:
The issue was that the business logic to update the trim settings was missing

The following [commit](https://github.com/Microchip-MPLAB-Harmony/csp/commit/1c254bbc9e697bbf60fd3d206196d6c3659aca83) resolves the issue.

macload1, 2019-08-26:
Excellent,
Thank you for your comment.

closed 2019-08-26 by aethaniel