Microchip-MPLAB-Harmony/csp #6: PIC32MX interrupt error for UART1-3

dchirgwin, 2019-12-06:
Attempts to use UART1, UART2, UART3 cause an error when generating the configuration, but only if the interrupts are enabled.
```
<FreeMarker>[Error]: An exception was thrown while attempting to markup a template file: D:\HarmonyFramework\csp\peripheral\evic_01166\templates\system\interrupts_vector_asm.h.ftl
<FreeMarker>[Error]: freemarker.core.InvalidReferenceException: The following has evaluated to null or missing:
==> .vars[INT_NUMBER]  [in template "interrupts_vector_asm.h.ftl" at line 34, column 56]
```
The underlying cause appears to be the way the interrupts are configured in the 01166 EVIC module.
In a few cases, interrupts are shared between peripherals, and the `evic.py` code understands this. It generates symbols like `EVIC_4_NUMBER`, `EVIC_5_NUMBER` for interrupts that are not shared, but for interrupts that are shared, it uses names like `EVIC_24_0_NUMBER`, `EVIC_24_1_NUMBER`, `EVIC_24_2_NUMBER`. See code around lines 332 and 392 in `csp\peripheral\evic_01166\config\evic.py`

The problem comes when this information is used by the template in `csp\peripheral\evic_01166\templates\system\interrupts_vector_asm.h.ftl`

The template assigns a local variable called `INT_NUMBER` with a format like `EVIC_4_NUMBER`. This works fine where interrupts are not shared, but where interrupts are shared, this fails. For example, there is no such variable as `EVIC_24_NUMBER`, because there are three variables for this interrupt `EVIC_24_0_NUMBER` etc.
This causes the lookup to fail, since the expression `.vars[INT_NUMBER]` fails when `INT_NUMBER` is   `EVIC_24_NUMBER`

As a temporary workaround, I have used UART4, which does not use a shared interrupt.

dsettu, 2019-12-06:
@dchirgwin , Thank you for your feedback and this issue will be fixed in the next release.

BTR1, 2020-01-13:
is there a work-around for this issue in current Harmony? That is: what script changes can be made to make it work?

dsettu, 2020-01-13:
@BTR1 , We will be releasing v3.6 this week and it will have fix for this issue. thanks.

dsettu, 2020-01-29:
@BTR1 , This issue is fixed now and released. Can you please check and close. thank you.

closed 2020-02-11 by aethaniel