Microchip-MPLAB-Harmony/csp #8: PIC32MX795F512L JTAG NOT Disable in SYS_Initialize Routuine

eastWillow, 2020-02-06:
### Problem
[image]
[image]

I didnt open the JTAG function but the GPIO is still not working.

schematic:
[image]
real:
[image]

### HOW to Resolve
I refer this forum post
https://www.microchip.com/forums/m734345.aspx
`DDPCONbits.JTAGEN = 0;`

Add the code:
[image]
then this will working
[image]

### Harmony 2 
I find in the Harmony 2 the code generator will generate the JTAG Disable Routine
[image]
`SYS_DEVCON_JTAGDisable();`

arpananand1, 2020-02-06:
thanks for reporting, we will fix this issue.

aethaniel, 2020-02-11:
@dsettu, please provide status here once fixed.

closed 2020-02-11 by aethaniel

arpananand1, 2020-03-06:
the issue has been fixed, it will be part of next release.