Microchip-MPLAB-Harmony/csp #12: Clock Configurator for e51 not working

jran-dasa, 2020-03-04:
Latest commit of clock configurator (clk_sam_e5x.jar) does not work, at least not with SAME51J19A.
Trying to start it from Harmony configurator does nothing.

Changed clk_sam_e5x.jar to previous version (from 26 July 2019) and it starts fine.
Using Harmony Configurator v3.4.1

aethaniel, 2020-03-04:
Hi @jran-dasa, thanks for reporting.
This has been converted as internal JIRA ticket to be addressed in next coming CSP v3.7.0 release within few weeks (MH3-30406).

dsettu, 2020-03-05:
@jran-dasa , I tried it with SAME51J19A using below repo version and clock configuration is launching.
[image]

Please check if you are using dev_packs v3.6.1 as clock configurator uses data from dev_packs.

jran-dasa, 2020-03-09:
@dsettu I am using the same set of csp, dev_packs and mhc versions as you are.

jran-dasa, 2020-03-09:
Exact steps I can use to reproduce:
1. Create new project in MPLAB X 5.30, Harmony 3 project for SAME51J19A
2. Launch Harmony Configurator (DFP 3.2.64 and CMSIS 5.6.0)
3. Tools->Clock Configuration, nothing happens.

I enabled all logging, and have attached log file.
[mhc.log](https://github.com/Microchip-MPLAB-Harmony/csp/files/4305116/mhc.log)

dsettu, 2020-06-08:
@jran-dasa , We fixed UI manager for handling locale issues. Please use v3.7.1 release.

closed 2020-06-08 by dsettu