Microchip-MPLAB-Harmony/csp #18: Support for SAM cortex-m3 MCUs

KhazAkar, 2020-06-22:
Hello, I'd want to ask - is it planned to add support for SAM Cortex-M3 MCUs like SAM3U or SAM3X series? They're great small MCUs with tons of capabilities but they can be seen now like little out of support.

syamknair, 2020-06-22:
Hi @KhazAkar,
As of now, we do not have any plans to support SAM Cortex-M3 MCUs in Harmony 3.

KhazAkar, 2020-06-22:
It's possible to know why? Just curious about future plans regarding SAM Cortex-M3 MCUs :)

syamknair, 2020-06-26:
Currently, the focus is on adding support for latest Microchip MCUs (with ARM cortex M0+, M4, M7, M23 cores and MIPS cores) in Harmony 3. You can continue using existing tools and framework for SAM Cortex-M3 MCUs from Microchip, if you decide to use them in your design.

closed 2020-06-26 by syamknair