Microchip-MPLAB-Harmony/csp #20: Bug in UART driver

dchirgwin, 2020-07-28:
A recent update has introduced a bug in the UART driver.
See commit 78230487672ab5bf943ae36c01006344ac0dd24d
https://github.com/Microchip-MPLAB-Harmony/csp/commit/78230487672ab5bf943ae36c01006344ac0dd24d

The problem was experienced on a PIC32MX device, using UART6 without a ring buffer,

The commit moved the point where the TX interrupt flag was cleared. Previously it was cleared after new data had been pushed into the TX FIFO, but this commit changed it to be cleared before new data had been pushed to the TX FIFO. Crucially, this means that the TX interrupt flag was cleared while the interrupt condition still existed (i.e. the TX buffer was still empty).

See the following note from the PIC32MX Technical Refernce manual Section 21.5.2

vishalnxt, 2020-07-28:
@dchirgwin , thank you for reporting the issue. We are looking into the issue and will get back.

vishalnxt, 2020-08-04:
@dchirgwin , This bug got introduced because of a fix done for an errata on PIC32MK devices. On PIC32MK, the transmit interrupt flag does not remain asserted when all characters are transmitted out. 

Consider the following scenario, where in the transmit interrupt, after filling the transmit FIFO and before clearing the transmit interrupt flag, another high priority interrupt occurs and the current interrupt execution gets preempted. When the CPU returns back to service the preempted transmit interrupt, all characters may have been transmitted out and the transmit FIFO can be empty. At this point, if the transmit interrupt flag is cleared, then on PIC32MK devices it does not remain asserted even though the condition for the interrupt flag is true. The suggested workaround for this is to clear the interrupt flag before writing to the transmit data register. 

However, on devices where the interrupt flag does remain asserted, clearing the interrupt flag while the interrupt condition is true results in interrupt flag getting set again, thereby firing another interrupt. However, as you rightly mentioned, this gets largely unnoticed because the ISR checks for the txBusyStatus flag and whether the transmit FIFO has space to accept a new character. 

Same applies in case of data reception. If the receive interrupt flag is cleared after reading out the receive FIFO but before receiving a new character in the FIFO (again due to another higher priority interrupt preempting the receive interrupt and thereby allowing enough time for a character to be received), then the receive flag remains cleared even though receive FIFO has unread characters.

Hope this provides the clarification.

The issue is fixed now by generating the workaround only for PIC32MK devices. The fix will be part of the next release - v3.8. Thank you again for reporting the issue.

syamknair, 2020-09-15:
Fixed as part of [v3.8.0](https://github.com/Microchip-MPLAB-Harmony/csp/releases/tag/v3.8.0)

closed 2020-09-15 by syamknair