Microchip-MPLAB-Harmony/csp #21: C++ compatibility of gpio functions

pustekuchen91, 2020-08-18:
peripheral/gpio/plib_gpio.h is missing some cast to be c++ compatible.

With this casts it can compile with xc32-g++
```
static inline void GPIO_PinWrite(GPIO_PIN pin, bool value)
{
    GPIO_PortWrite((GPIO_PORT)(pin >> 4), (uint32_t)(0x1) << (pin & 0xF),
                   (uint32_t)(value) << (pin & 0xF));
}

static inline bool GPIO_PinRead(GPIO_PIN pin)
{
    return (bool)(((GPIO_PortRead((GPIO_PORT)(pin >> 4))) >> (pin & 0xF)) & 0x1);
}

static inline bool GPIO_PinLatchRead(GPIO_PIN pin)
{
    return (bool)((GPIO_PortLatchRead((GPIO_PORT)(pin >> 4)) >> (pin & 0xF)) & 0x1);
}

static inline void GPIO_PinToggle(GPIO_PIN pin)
{
    GPIO_PortToggle((GPIO_PORT)(pin >> 4), 0x1 << (pin & 0xF));
}

static inline void GPIO_PinSet(GPIO_PIN pin)
{
    GPIO_PortSet((GPIO_PORT)(pin >> 4), 0x1 << (pin & 0xF));
}

static inline void GPIO_PinClear(GPIO_PIN pin)
{
    GPIO_PortClear((GPIO_PORT)(pin >> 4), 0x1 << (pin & 0xF));
}

static inline void GPIO_PinInputEnable(GPIO_PIN pin)
{
    GPIO_PortInputEnable((GPIO_PORT)(pin >> 4), 0x1 << (pin & 0xF));
}

static inline void GPIO_PinOutputEnable(GPIO_PIN pin)
{
    GPIO_PortOutputEnable((GPIO_PORT)(pin >> 4), 0x1 << (pin & 0xF));
}
```

jigneshmoradiya1, 2020-08-18:
@pustekuchen91, Thanks for reporting the issue. The fix will be available in the next H3 csp v3.8.0 release.

syamknair, 2020-09-15:
Fix available as part of [v3.8.0](https://github.com/Microchip-MPLAB-Harmony/csp/releases/tag/v3.8.0)

closed 2020-09-15 by syamknair