Microchip-MPLAB-Harmony/csp #22: AD1CSSL Value gets overwritten

pustekuchen91, 2020-09-17:
Hello,

sometimes when regenerating all harmony files, the value of the AD1CSSL Register will be overwritten with another Value.

I'm using the following configuration for the adc.
[image]

It seems that the configurator doesnt recognize the choosen Input Scan Pins.

PIC32MX450F256H
CSP 3.8.1

pustekuchen91, 2020-09-17:
After restarting MHC and regenerating he tries to change the value again.
[image]

arpananand1, 2020-09-17:
@pustekuchen91 is it possible to zip and share the complete mplab project?

dsettu, 2020-09-17:
@pustekuchen91 , I see that AN2, AN3, AN4 and AN5 channels are enabled for channel scan. It should result in AD1CSSL value of 0x3C. Not sure why the earlier code has 0x38 for AD1CSSL register.

pustekuchen91, 2020-09-18:
@dsettu your right AN2-5 are enabled and should result in value of 0x3C. This value is already set at the current code (current code = right side). 

MHC wants to replace with 0x38 (left side) (where AN2 is not set).

Maybe i can create a new project next week.

dsettu, 2020-09-18:
@pustekuchen91 , Please let us know if you see this behavior in new project. thanks.

pustekuchen91, 2020-09-21:
Hello @dsettu 

i've created a brand new project and configured the adc and generated the harmony files. After that i've closed the MHC and reopened and recreated the files. Merge Strategy: All -> The following changes wants to be made:
[image]

so the problem is 100% reproducible. I've attached the created project.

[ADC_Bug.zip](https://github.com/Microchip-MPLAB-Harmony/csp/files/5253933/ADC_Bug.zip)

prachiap, 2020-09-21:
@pustekuchen91 , I can reproduce this issue. AD1CSSL Register value is not calculated correctly. 
It will be fixed in the next release. 

Thanks for reporting this issue.

prachiap, 2020-09-21:
@pustekuchen91 , I have attached the modified file. Please copy/replace this python file at location:
\csp\peripheral\adc_00755\config

[adc.zip](https://github.com/Microchip-MPLAB-Harmony/csp/files/5256038/adc.zip)

Let us know if this solves the issue.

pustekuchen91, 2020-09-22:
@prachiap thanks, this fix works! 

Maybe you should take a look at AD1CON3, too. As you can see in the last screenshot of the demo project. AD1CON3 was not included at the initial generation and was added at the second run.

prachiap, 2020-09-22:
@pustekuchen91 , AD1CON3 register will be generated when the AD1CON3 value is non-zero. Once you save the configurations, AD1CON3 value is getting saved correctly. I do not see that,  this difference is showing again and again. 

Let me know if you see any mismatch in this register value.

pustekuchen91, 2020-09-22:
@prachiap your right, the difference is not showing again and again. But AD1CON3 was missing at the first generation of the harmony files.

- Create new Project
- Configure as following 
[image]

After first generation: 

`void ADC_Initialize(void)
{
    AD1CON1CLR = _AD1CON1_ON_MASK;

    AD1CON1 = 0xe4;

    /* Turn ON ADC */
    AD1CON1SET = _AD1CON1_ON_MASK;
}`

as you can see AD1CON3 is missing. I've added a zip file with the xml files after first and second generation. as you can see 
[default.zip](https://github.com/Microchip-MPLAB-Harmony/csp/files/5259630/default.zip)

this part is missing in xml file
[image]

prachiap, 2020-09-23:
@pustekuchen91 , Thanks for the details. 
AD1CON3 value is not getting calculated for the first time. 
Please find attached updated file. Copy it to \csp\peripheral\adc_00755\config

[adc_AD1CON3_fix.zip](https://github.com/Microchip-MPLAB-Harmony/csp/files/5266459/adc_AD1CON3_fix.zip)

pustekuchen91, 2020-09-23:
@prachiap works for me. thanks for the fix.

closed 2020-09-23 by pustekuchen91