Microchip-MPLAB-Harmony/csp #23: [EVIC] interrupts_vector_table.h.ftl error

pustekuchen91, 2020-09-29:
PIC32MX795F512L

Error on generating files:
```
<FreeMarker>[Error]: An exception was thrown while attempting to markup a template file: C:\microchip\harmony\v3\csp\peripheral\evic_01166\templates\system\interrupts_vector_table.h.ftl
<FreeMarker>[Error]: freemarker.core.InvalidReferenceException: The following has evaluated to null or missing:
==> .vars["EVIC_PRIORITY_" + .vars[INT_PRIORITY] + "ATTRIBUTE"]  [in template "interrupts_vector_table.h.ftl" at line 81, column 82]

----
Tip: It's the final [] step that caused this error, not those before it.
----
Tip: If the failing expression is known to be legally refer to something that's sometimes null or missing, either specify a default value like myOptionalVar!myDefault, or use <#if myOptionalVar??>when-present<#else>when-missing</#if>. (These only cover the last step of the expression; to cover the whole expression, use parenthesis: (myOptionalVar.foo)!myDefault, (myOptionalVar.foo)??
----

----
FTL stack trace ("~" means nesting-related):
	- Failed at: ${.vars["EVIC_PRIORITY_" + .vars[INT_...  [in template "interrupts_vector_table.h.ftl" at line 81, column 80]
----
<FreeMarker>[Error]: 	- Failed at: ${.vars["EVIC_PRIORITY_" + .vars[INT_...  [in template "interrupts_vector_table.h.ftl" at line 81, column 80]
<Generate>[Info]: An exception occured during file preprocessing.
<Log>[Error]: Exception cause: java.lang.Exception: Error during file preprocessing.
<Log>[Error]: Stack trace:
<Log>[Error]: com.microchip.mh3.generate.GenerateCommon.preprocessFileToString(GenerateCommon.java:93)
<Log>[Error]: com.microchip.mh3.generate.engine.mplabx.GenerateEngineMPLABX.generate(GenerateEngineMPLABX.java:128)
<Log>[Error]: com.microchip.mh3.generate.Generate$2.run(Generate.java:361)
<Log>[Error]: java.lang.Thread.run(Thread.java:748)
<Generate>[Error]: Aborting project generation.  Your project has not been modified.
```

if all changes made in 2020 which effects [EVIC] are revered, the file generation is successful.

https://github.com/Microchip-MPLAB-Harmony/csp/blob/88639cb69340ec3a2d68ef385d99adaeffd8eaba/peripheral/evic_02907/templates/system/interrupts_vector_table.h.ftl#L35

arpananand1, 2020-09-29:
are you getting this issue with a new project you are trying to create or some old project which you are trying to regenerate with latest version?

arpananand1, 2020-09-29:
make sure you are using latest csp\peripheral\evic_02907\config\evic.py file as well.

pustekuchen91, 2020-09-29:
it is a project which was created with an older version of csp. 
evic.py should be up to date, i just made a git pull to the newest version
I've readded Core Module to the Project Graph (which helps with some modules), with no effect.

arpananand1, 2020-09-29:
are you using latest "dev_packs" repo, without that such issue will occur.

pustekuchen91, 2020-09-29:
I wasnt. I've updated `dev_packs` to `1.4.241`. The error still remains.

arpananand1, 2020-09-29:
i suppose you have closed the mplab project and reopened after updating dev_packs?
if issue is still persisting, is it possible to share the project?

also, can you try creating a new project and see if issue comes there or not?

pustekuchen91, 2020-09-30:
thank you, it works now.

closed 2020-09-30 by pustekuchen91