Microchip-MPLAB-Harmony/csp #25: SAM D5x/E5x Family 16-bit Timer data structure issue

kick2nick, 2020-12-30:
Hello,
Generated tc.h file conatain tc_count16_registers_t structure
This structure has 16-bit long fields "tc_cc".
According to datasheet these fields have 16-bit width.
But in fact it is 32-bit width fields(MPlab IO View and hardware tests with PWM mode).
So we have mistakes in datasheet: addresses and sizes of CC0,CC11,CCBUF0,CCBUF1 for 16-bit TC mode. 
I didn't find any information about this problem in Errata sheet.
Also we have mistakes in Harmony generated code according to datasheet.
Best regards,
Nikolay
[image]
[image]
[image]
[image]

prachiap, 2021-01-18:
Hi Nikolay, 

CC, PER Register widths are different for 8-bit/16-bit/32-bit mode in TC. 
In datasheet also, we have different sections and offset given for different modes. 

Do you see any discrepancy in writing to CC or PER registers? 
Do you see any functional issue with period and duty cycle of the generated waveform?

Thanks,
Prachi

[image]

[image]

[image]

kick2nick, 2021-01-21:
Generated code gives me PWM with one clock positive pulse(I set 50% duty in GUI).
When I'm trying to set new value it has no effect.
Pulse duration should be set by CC1 register, but it doesn't change.(see pic. 1)
I noticed that in IO view registers CC0 CC1 have 32-bit widths(for 16-bit timer), so I tried to use 32-bit pointer and it worked.(see pic. 2)
Look at IO view: CC0 addr - 0x1C, CC1 addr -0x20. I don't see such addresses in register summary for all timers.

prachiap, 2021-01-21:
Based on the code snippet, I believe that you are using MPWM mode to generate PWM waveform. 
In MPWM waveform, CC0 decides period and CC1 decides duty cycle and output should be observed on WO[1] pin. WO[0] output is one always clock pulse irrespective of CC1 value. 

Can you please let me know:
1. Which device are you using?
2. Are you checking WO[1] output?
3. Is waveform frequency coming correctly?

Regards,
Prachi

prachiap, 2021-01-25:
Hi Nikolay,

Are you able to get correct period and duty on WO[1] pin?
Please let me know if issue still persists. 

Regards,
Prachi

vishalnxt, 2021-02-15:
@kick2nick , I am closing the ticket as there is no update from your side. If you are still facing the issue, you can reopen the ticket and provide your observations. Thank you.

closed 2021-02-15 by vishalnxt