Microchip-MPLAB-Harmony/csp #27: UART Polling process problem

khlee11, 2021-02-10:
Device: PIC32MK MCJ Curiosity Pro EVB
IDE: MPLAB X IDE v5.40, XC32 v2.50
BootLoader: Harmony3 UART_Bootloader (https://github.com/Microchip-MPLAB-Harmony/bootloader_apps_uart/tree/master/apps/uart_bootloader)

I have other questions.
I am writing UART transmission/reception by polling method.
Transmission seems to be fine in PIC.
However, there is a problem in receiving processing.
It appears to be a phenomenon caused by RX FIFO on the UART module of PIC32MK.
RX FIFO is 8-Level and up to 8 bytes can be received without problem.
However, if 9 bytes are transmitted, reception processing is performed at the first time, but reception processing is not performed at all after that.

I try to receive 8 bytes 3 times, and finally, after receiving 9 bytes, I cannot receive it.
[image]

What causes this phenomenon?
No matter how simple you modify the code, it is the same.
The code I received and processed is as follows.
[image]

vishalnxt, 2021-02-15:
@khlee11 , Can you attach the MPLABX project for us to debug the issue?

khlee11, 2021-02-15:
Attach the requested project file.
It is simply a code that receives and returns the data sent from the host.
[firmware_210215.zip](https://github.com/Microchip-MPLAB-Harmony/csp/files/5979643/firmware_210215.zip)

khlee11, 2021-02-15:
Attach the requested project file.
It is simply a code that receives and returns the data sent from the host.
[Uploading firmware_210215.zip…]()

vishalnxt, 2021-02-15:
@khlee11 , thank you for the response, but I am not able to download the zip file you shared. It seems to be pointing back to the issue. Can you please check. 
Meanwhile, did you try the UART Echo Blocking demo available here - [uart_echo_blocking](https://github.com/Microchip-MPLAB-Harmony/csp_apps_pic32mk/tree/master/apps/uart/uart_echo_blocking/firmware)

khlee11, 2021-02-15:
> @ khlee11 님 , 응답 해 주셔서 감사합니다. 공유하신 zip 파일을 다운로드 할 수 없습니다. 문제를 다시 지적하는 것 같습니다. 확인해주세요.
> 한편, 여기에서 제공되는 UART Echo Blocking 데모를 사용해 [보셨습니까-uart_echo_blocking](https://github.com/Microchip-MPLAB-Harmony/csp_apps_pic32mk/tree/master/apps/uart/uart_echo_blocking/firmware)

[firmware_210215.zip](https://github.com/Microchip-MPLAB-Harmony/csp/files/5979710/firmware_210215.zip)

vishalnxt, 2021-02-15:
@khlee11 , can you remove the printf calls and try out the below code :

`void uart_rx(void)
{
    uint8_t data;
    
    if(UART1_ReceiverIsReady() == true)
    {
        if(UART1_ErrorGet() == UART_ERROR_NONE)
        {
            data = UART1_ReadByte();
            UART1_WriteByte(data);
        }
    }
}

int main ( void )
{

    /* Initialize all modules */
    SYS_Initialize ( NULL );

    while ( true )
    {
        uart_rx();
    }

    /* Execution should not come here during normal operation */

    return ( EXIT_FAILURE );
}
`

khlee11, 2021-02-15:
I tried with the code you suggested.
However, the symptoms are the same.
It's a very simple operation, but I think I'm missing something simple.
Data is received only as deep as the FIFO.

vishalnxt, 2021-02-19:
@khlee11 , are you still seeing this issue? I have uploaded a simple UART blocking demo, which we tested at 115200 baud rate and by typing characters on a UART console (Tera Term). Can you check if that works for you?

[uart_echo_blocking_pic32mk_mcj_curiosity_pro.zip](https://github.com/Microchip-MPLAB-Harmony/csp/files/6008768/uart_echo_blocking_pic32mk_mcj_curiosity_pro.zip)

Moreover, in the project that you shared, I notice that the STDIO in the project graph is not connected to any UART. You need to connect it to UART1. Also the UART pins seem to be configured incorrectly. The right UART pins for the PIC32MK MCJ Curiosity board are: U1RX:RG8, U1TX:RE0

Can you make these changes and see if that helps?

khlee11, 2021-02-22:
Thanks for the answer.
I have identified the problem.
We confirmed this as a problem caused by CoreTimer_ Delay in the bootloader switch function.
FIFO overflow seems to occur when the delay function is performed.
When I removed the delay, it worked normally.
I did not connect UART to STDIO because I wrote custom code to selectively apply UART1 and UART2, so I did not connect.
Thanks again for the answer.

closed 2021-02-22 by khlee11