Microchip-MPLAB-Harmony/csp #28: printf processing issue

khlee11, 2021-02-18:
Device: PIC32MK MCJ Curiosity Pro EVB
IDE: MPLAB X IDE v5.40, XC32 v2.50
BootLoader: Harmony3 UART_Bootloader (https://github.com/Microchip-MPLAB-Harmony/bootloader_apps_uart/tree/master/apps/uart_bootloader)

I am doing UART transmission by connecting printf.
Most of the communication is performed normally.
However, the data of certain characters is a problem.
I am sending a string.
It contains a mixture of ASCII and hexadecimal characters.
I sent the data "5B 0A 43 58 45 4E 41 42 4C 45 8F 5D".
  printf("[%cCXENABLE%c]",0x0A10,0x8F);  
[image]

I wrote it with the code above.
When the code is executed, data is transmitted,
Data on the receiving side is received as "5B 0D 0A 43 58 45 4E 41 42 4C 45 8F 5D".
[image]

If 0A (ASCII-Line Feed) is always sent, 0D (ASCII-Carriage Return) is attached in front of 0A and received in the form of 0D 0A.
It seems to change that way in printf processing.
I am curious about the cause of this phenomenon and how to improve it.

vishalnxt, 2021-02-25:
@khlee11 the implementation of printf expands Line Feed (0x0A) to Carriage Return + Line Feed (0x0D + 0x0A). Moreover, the printf function tries to print in a readable/text format. For example, when using printf("%x", 0xAB), it prints/transmits two ascii characters - 'a' (0x61) and 'b' (0x62), so that it is displayed on the terminal as 'ab'. In my view, there is no straight forward way to transmit binary data using printf function.

khlee11, 2021-02-25:
Thank you.
I have to find another way.

closed 2021-02-25 by khlee11