Microchip-MPLAB-Harmony/csp #29: NVM Driver Documentation missing function

michaelidesmi, 2021-05-13:
The Library interface documentation for the NVM driver is missing the NVM_QuadWordWrite entry.
Quad Word write is available on PIC32MZ devices and is it already implemented in the harmony driver but not listed in the documentation.

arpananand1, 2021-05-13:
are you using the latest doc? i can see the mentioned API documented
[image]

michaelidesmi, 2021-05-13:
You are right! I downloaded the CSP again and its there. Sorry for the confusion

closed 2021-05-13 by michaelidesmi

arpananand1, 2021-05-13:
cool. BTW i noticed small mistake in documentation in the return type of the API. we shall be fixing it in next release. FYI