Microchip-MPLAB-Harmony/csp #30: SPI Library Documentation

michaelidesmi, 2021-06-04:
I've noticed 2 minor documentation issues for the SPI Peripheral Library. I am using a PIC32MZ.

In the **SPIx_WriteRead** documentation it is stated that _It returns False if either of the size parameters are 0 and corresponding data pointer is NULL._ which is not accurate. The function will return false if both size parameters are 0.

In the **SPIx_TransferSetup** there is no mention of the fact that for the **spiSourceClock** a value of 0 will default to _Use Master Clock Frequency set in GUI_. If this is something that will become obsolete in the future ignore it.

vishalnxt, 2021-06-21:
@michaelidesmi , thank you for bringing the issues to our notice. We will update the documentation in the next release.

vishalnxt, 2021-09-23:
@michaelidesmi , the issue is fixed in v3.10 release. I am closing the ticket. Feel free to re-open it in case you find any issues.

closed 2021-09-23 by vishalnxt