Microchip-MPLAB-Harmony/csp #31: 2 CANFD on PIC32MK1024MCM100

raf6410, 2021-06-14:
Hi,

I work on PIC32MK1024MCM100 an CANFD in a custum board.
All CAN channel work correctly alone.
But when I tried to use 2, 3 or 4 CAN in same time PIC CRASH.
I think that the PIC don't start.

Could you help me 

RAF6410

[image]

jigneshmoradiya1, 2021-06-14:
Hi,

Data Bit Timing is not correctly configured for 4000 Kbps Bit Rate because Calculated Bit Rate is 0 and Error % is 100%. CAN-FD Clock Frequency should be configured 20MHz/40MHz/80MHz as per datasheet (https://ww1.microchip.com/downloads/en/DeviceDoc/PIC32MK-GPK-MCM-with-CAN-FD-Family-DataSheet-DS60001519E.pdf, Page#463).
For example, Please find the MHC configuration for CAN-FD clock frequency 20MHz, Nominal Bit Rate 500 Kbps and Data Bit Rate 4000 Kbps as below.
[image]

Thank you.

raf6410, 2021-06-16:
hi,

Thanks for your comment.
I tried to config 2 CAN channel with your proposal => same probleme, PIC don't start.
I use REFCLK4 to 20Mhz but nothing

raf6410, 2021-06-16:
After many tried, the problem appear only when i add CAN2.
Actually my PIC works with CAN1, CAN3, CAN4 and UART2 ON.

jigneshmoradiya1, 2021-09-28:
@raf6410 , CAN Bit Timing should be configured in MHC configuration as mentioned in above screenshot. Closing the ticket now. In case of any issue please feel free to reopen ticket.

closed 2021-09-28 by vishalnxt