Microchip-MPLAB-Harmony/csp #36: Vector Table incorrectly defined

CharleO, 2021-10-04:
For the ATSAMC20 and ATSAMC21 (Cortex M0+) there are 32 interrupts, after PTC_Handler there should be reserved vector. The header file "device_vectors.h" should have `pfn_handler_t pfnReserved31;` after `pfn_handler_t pfnPTC_Handler;` so that it falls on a page boundary.

vishalnxt, 2022-04-20:
@CharleO , May I know what issue you are running into with the existing implementation?

CharleO, 2022-05-16:
So my application Vector Table Offset Register is assigned to 0x00000800 address, where the boot loader wrote the application.  The Application Vector Table size includes first 16 exception and device's IRQ, which is 192 bytes.   Page boundary is 64 bytes. If all 32 IRQ are not defined properly, problems with validations tests.  Since this writing, I would also include the PIC32CM family as well.

vishalnxt, 2023-06-22:
@CharleO , I am closing this ticket as this is something that needs to be taken care in your validation tests.

closed 2023-06-22 by vishalnxt