Microchip-MPLAB-Harmony/csp #40: Harmony Configurator's Pin Configuration and Peripherals availability for SAMD10C14A differs from datasheet

nicotrozzo, 2021-12-27:
Hi,
I'm trying to use SERCOM2 with SAMD10C14's pins 4 and 5 (PA14 and PA15), since the datasheet says they are available as SERCOM-ALT:
[image]
However, I have two problems when using the MPLAB Harmony Configurator:
- SERCOM2 isn't available as a Peripheral
- SERCOM2 isn't available as an alternative for these pins
[image]
Maybe I'm getting something wrong from the docs or there is a reason why it's not available, I'm sorry if that's the case, but I couldn't find any of these in the documentation.

vishalnxt, 2022-01-06:
@nicotrozzo , thank you for reporting the issue. We are looking into it and will get back to you soon.

vishalnxt, 2022-04-12:
@nicotrozzo , the SAM D10C 14-pin variant does not have SERCOM2 (It only has SERCOM0 and SERCOM1). The SERCOM2 is available only on SAM D10D variants. Please refer to the configuration summary chapter in the datasheet. Based on the configuration summary, the SERCOM2 configuration should be ignored for the SAM D10C 14 pin variant. Hope this clarifies. 
[image]

vishalnxt, 2022-04-20:
@nicotrozzo , Since there is no further update from you I am closing this issue. Feel free to reopen it in case you have any further questions.

closed 2022-04-20 by vishalnxt

nicotrozzo, 2022-04-20:
Yes it's fine, thanks! I'd say it's an issue for the MCU docs, and actually Harmony helped me realize this problem.