Microchip-MPLAB-Harmony/csp #41: PWM0_CH3 not listed in configuration for ATSAME70N19B

ek-ap, 2022-01-05:
For an ATSAME70N19B (LQFP100 package), the PWM0 module configuration in Harmony 3 shows only channels 0..2, but not channel 3. For the PWM1 module, all four channels 0..3 are shown as expected.

**Analysis**
The ATDF file ` .\dev_packs\Microchip\SAME70_DFP\4.6.98\same70b\atdf\ATSAME70N19B.atdf` lists the following signals for PWM0 with group "PWMH"
```
<module name="PWM" id="6343" version="Y">
 <instance name="PWM0">
  <signals>
   <signal pad="PA7" function="B" group="PWMH" index="3"/>
   <signal pad="PA14" function="B" group="PWMH" index="3"/>
   <signal pad="PA17" function="C" group="PWMH" index="3"/>
   <signal pad="PD23" function="A" group="PWMH" index="3"/>
 </signals>
[...]
```
The `pinouts` section for the LQFP100 package lists PA7, PA14, and PA17, but not PD23. This seems in line with the data sheet. 

Considering the following code, where it is checked if the output pins for the individual PWM channels are available:
https://github.com/Microchip-MPLAB-Harmony/csp/blob/6048f901818171f60792a0cd228a4e4d1bbeb428/peripheral/pwm_6343/config/pwm.py#L497-L507

Here, the last `padSignal` encountered is PD23, which is not available on the LQFP100 package. However, this should not matter as three other pins are available for the PWM channel 3 signal. 

It seems, the code silently assumes that each PWMH signal may be assigned to at most one output pin. However, with the situation as described herein, the logic to my understanding should be "display channel configuration if there is _at least one_ output pin available". As the `channel` array is initialized to `False` in line 481, it should suffice to remove lines 506-507:

```
   for pad in range(0, len(pwm_signals)):
        if "PWMH" in pwm_signals[pad].getAttribute("group"):
            padSignal = pwm_signals[pad].getAttribute("pad")
            if padSignal in availablePins:
                channel[int(pwm_signals[pad].getAttribute("index"))] = True
            #else:
            #    channel[int(pwm_signals[pad].getAttribute("index"))] = False
```
This fixed the issue on my project.

vishalnxt, 2022-01-20:
@ek-ap , thank you for reporting the issue. We are looking into it and will get back.

dsettu, 2022-07-19:
Resolved as part of csp v3.13 release

closed 2022-07-19 by dsettu