Microchip-MPLAB-Harmony/csp #42: SMBus features request for PIC32MZ series devices.

youpko, 2022-02-21:
I am trying to implement SMBus logic with the I2C driver for a PIC32MZ2048EFM100. 
My application uses the above mention PIC MCU but this feature request is probably mostly the same for other MCUs in the PIC32MZ EF series.

The MCU should be an SMBus Master & Slave . 
 - Current harmony only allows me to select either a I2C Master or I2C Slave implementation not both.

SMBus BlockRead & BlockWrite-BlockRead not supported. 
- The BlockRead & BlockWrite-BlockRead Protocols specify that the first byte read specifies the number of bytes that follow and the current driver doesn't support this dynamical reading capability.
- All other SMBus BUS Protocols I can make with the available driver functions.

I checked "PIC32 Family Reference Manual, Sect. 24 Inter-Integrated Circuit" manual and in the overview it gives an example that this is possible as the I2C Master & I2C Slave work independently form each other in the I2C Module.

As I am not to familiar with the internal workings of the Harmony project I didn't manage to implement this myself yet, hence my request to look if this can be integrated in the driver for the devices that support is.

dsettu, 2022-07-19:
@youpko , We have logged the request to add support for SMbus

agen007niaga, 2022-09-02:
ejen007niaga

vishalnxt, 2024-12-05:
The SMBUS support for PIC32M devices is available with CSP v3.20.0. Hence, closing this ticket. Feel free to reopen the ticket if you face any issues.

closed 2024-12-05 by vishalnxt