Microchip-MPLAB-Harmony/csp #54: Extended CAN template issue

arunningpir8, 2022-08-24:
https://github.com/Microchip-MPLAB-Harmony/csp/blob/6b191b04015eaf9446280b2f81a3b88368c8f9a3/peripheral/can_01152/templates/plib_can.c.ftl#L234

should read:
`txMessage->msgEID = ((id & 0x3FFFF) << 10) | CAN_MSG_IDE_MASK | CAN_MSG_SRR_MASK;`

this causes many issues in Extended CAN applications.

As per Microchip documentation:
SRR: Substitute Remote Request bit
In case of a standard message format (IDE = 0), this bit is don’t care.
In case of an extended message format (IDE = 1), this bit should always be set.

jigneshmoradiya1, 2022-08-24:
@arunningpir8 , Thanks for reporting this issue. This issue will be fixed in the next csp release.

vishalnxt, 2023-06-22:
@arunningpir8 , this issue is resolved in the latest CSP release v3.17.0. Hence, we are closing this issue. Feel free to reopen if you are still facing any problem.

closed 2023-06-22 by vishalnxt