Microchip-MPLAB-Harmony/csp #69: PIC32MZW WTD reset clock initialisation

danwood76, 2023-04-12:
The following change causes the device to lockup after a WDT reset occurs as the clocks are no longer initialised:
https://github.com/Microchip-MPLAB-Harmony/csp/commit/38e536e3c657499b2fba0bc6cb63c9be78e9bf80

Adding a test for the "RCONbits.WDTO" bit as with the "RCONbits.SWR" bit works around the issue.

aditya-shankar, 2023-04-13:
Hi,

This code has been internally modified(not yet published though) to remove the multiple RCON register checks we had because it proved to be increasingly troublesome with the amount of combinations with which the boot code was invoked.

This change has been through few internal checks and proved to hold up well so far, so you might want to try this out at your end as well and let me know if this solves your problem. 

[plib_clk_status_check.patch](https://github.com/Microchip-MPLAB-Harmony/csp/files/11218433/plib_clk_status_check.patch)

Thanks,
Aditya

danwood76, 2023-04-13:
Hi,

Thanks for the quick response and the patch.

I can confirm that with this patch applied I no longer get a hang after a WDT reset.

Kind regards,
Danny

aditya-shankar, 2023-04-14:
Hi Danny,

Glad to hear that!
Thanks for your confirmation.

Regards,
Aditya

closed 2023-06-22 by vishalnxt