Microchip-MPLAB-Harmony/csp #74: Pins Missing in 3.19.2

arunningpir8, 2024-07-10:
I just lost my mind for 3 days trying to start a new project using the newest 3.19.2 CSP before finally realizing this version CSP was responsible for pins not showing up on my ATSAMV71N21B.

To replicate:
Set CSP in Harmony to 3.19.2 with a project using ATSAMV71N21B
Open Pin Manager
Note that about 1/3 of the pins are missing from the tables and associated graphs

Switch back to 3.19.1
Open Pin Manager
All pins show up for configuration again

Also note, that if you attempt to use 3.19.2 and setup pins that DO show up, they get mapped to completely wrong pins in pin_configuration.csv as well, making this release completely broken

jigneshmoradiya1, 2024-07-11:
@arunningpir8 , Thanks for reporting this issue. This Pin Manager issue is fixed in the latest csp v3.19.3 release (https://github.com/Microchip-MPLAB-Harmony/csp/tree/v3.19.3)

arunningpir8, 2024-07-11:
Thanks for the fast update!  I can confirm things are back to working with the 3.19.3 release!

closed 2024-07-11 by arunningpir8