Microchip-MPLAB-Harmony/csp #76: TSENS peripheral missing

Odry01, 2024-10-01:
Hi, I want to use TSENS peripheral in MPLAB MCC Harmony project with SAMC21E18A microcontroller. When I followed the course at Microchip University and saw in the chapter TSENS - MCC Configuration this:

[image]

but when I want to use this peripheral in my project, I only have this:

[image]

I don't have this peripheral in the peripherals menu, yet in the video course and used. I also tried to create a project with the same microcontroller as in the video, but unfortunately I did not find it there either. It wasn't even in the list of peripherals when I looked here on GitHub in the "peripheral" folder. I have MCC v5.5.1 and MPLAB 6.20 installed with the latest csp update - 3.19.6. Why don't I have the TSENS peripheral available and is there any chance it will be back in a future update?

jigneshmoradiya1, 2024-11-26:
@Odry01 , Thanks for reporting this issue. TSENS PLIB will be available in the next csp v3.20.0 release.

Odry01, 2024-11-26:
Thank you very much

closed 2024-11-27 by Odry01