Microchip-MPLAB-Harmony/csp #77: Generating uart plib files for same5 does not work anymore

Psy-Kai, 2025-01-28:
Hello,

[a9e29f9b4144366908e4f0fbe53c10ba5f3d91b9 ](https://github.com/Microchip-MPLAB-Harmony/csp/commit/a9e29f9b4144366908e4f0fbe53c10ba5f3d91b9) introduced `LIST_SERCOM_7816_C` which leads to an exception while generating:

```
FreeMarker template error:
The following has evaluated to null or missing:
==> LIST_SERCOM_7816_H  [in template "plib_sercom5_usart.h" at line 130, column 3]

----
Tip: If the failing expression is known to be legally refer to something that's sometimes null or missing, either specify a default value like myOptionalVar!myDefault, or use <#if myOptionalVar??>when-present<#else>when-missing</#if>. (These only cover the last step of the expression; to cover the whole expression, use parenthesis: (myOptionalVar.foo)!myDefault, (myOptionalVar.foo)??
----

----
FTL stack trace ("~" means nesting-related):
	- Failed at: ${LIST_SERCOM_7816_H}  [in template "plib_sercom5_usart.h" at line 130, column 1]
----

Java stack trace (for programmers):
----
freemarker.core.InvalidReferenceException: [... Exception message was already printed; see it above ...]
	at freemarker.core.InvalidReferenceException.getInstance(InvalidReferenceException.java:131)
	at freemarker.core.EvalUtil.coerceModelToString(EvalUtil.java:355)
	at freemarker.core.Expression.evalAndCoerceToString(Expression.java:82)
	at freemarker.core.DollarVariable.accept(DollarVariable.java:41)
	at freemarker.core.Environment.visit(Environment.java:324)
	at freemarker.core.MixedContent.accept(MixedContent.java:54)
	at freemarker.core.Environment.visit(Environment.java:324)
	at freemarker.core.Environment.process(Environment.java:302)
	at freemarker.template.Template.process(Template.java:325)
	at com.microchip.mcc.harmony.HarmonyModuleGenerator$3$1.render(HarmonyModuleGenerator.java:493)
	at com.microchip.mcc.core.service.BaseGenerationService.generateItem(BaseGenerationService.java:721)
	at com.microchip.mcc.core.service.BaseGenerationService$MergeDecision.generate(BaseGenerationService.java:633)
	at com.microchip.mcc.core.service.BaseGenerationService$MergeDecision.access$200(BaseGenerationService.java:531)
	at com.microchip.mcc.core.service.BaseGenerationService.lambda$generateFiles$0(BaseGenerationService.java:218)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:750)
Caused: com.microchip.mcc.core.generation.MCCGenerationException: Generation error:render
	at com.microchip.mcc.harmony.HarmonyModuleGenerator$3$1.render(HarmonyModuleGenerator.java:496)
	at com.microchip.mcc.core.service.BaseGenerationService.generateItem(BaseGenerationService.java:721)
[catch] at com.microchip.mcc.core.service.BaseGenerationService$MergeDecision.generate(BaseGenerationService.java:633)
	at com.microchip.mcc.core.service.BaseGenerationService$MergeDecision.access$200(BaseGenerationService.java:531)
	at com.microchip.mcc.core.service.BaseGenerationService.lambda$generateFiles$0(BaseGenerationService.java:218)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:750)

```

This is one of many errors I encountered (even in `core` module). We are currently not able to generate a any sane project via harmony code configurator!

vishalnxt, 2025-01-28:
@Psy-Kai , thanks for reporting the issue. We are looking into it and will get back with more details.

Psy-Kai, 2025-01-28:
Thanks. For your interest: I got the messages from 
* MPLAB X (assuming the default window layout)
* opening the MCC
* generating files
* switch to the notification tab at the message pane at the bottom
* click on "Unexpected Exception"
* "Show and Report Problem"

There I got about 10 Templare-File problems.

SumanthSuraneni, 2025-01-28:
@Psy-Kai , We are unable to reproduce this issue on our end. If you feel comfortable, kindly share the project zip file with us so we can investigate further.   

Kindly close the MPLAB X IDE and reopen it to clear the log file data. Open the project, launch MCC, and reproduce the issue. Please share the MCC log files with us.

vishalnxt, 2025-01-28:
@Psy-Kai mcc log is usually available under C:\Users\user_name\.mcc\log

Psy-Kai, 2025-01-31:
Ok, installing the xc32 resolves the problem... But will close the issue.

But I am still wondering, why I need the xc32 installed to generate the files...

closed 2025-01-31 by Psy-Kai